fprime-community/fprime-tutorial-arduino-blinker #10: Encounting many issues while regenerating

aggelosmots, 2024-03-11:
I'm trying to regenerate the arduino example using a Teensy 4.0. Firstly what I did was to clone the repo and see if it builds normally. Everything is fine until I try to make things by my own.

I'm using 3.4.1 version of fprime.

Issues:
1) Cookie cutter error:   when I add `deployment_cookiecutter: https://github.com/fprime-community/fprime-arduino-deployment-cookiecutter.git` I get the error 
```
[ERROR] expected token ':', got ','
  File "Top/{{cookiecutter.deployment_name}}Topology.cpp", line 28
    Svc::RateGroupDriver::DividerSet rateGroupDivisors = {{ {100, 0}, {200, 0}, {1000, 0} }};
```

2) SPI and Wire libraries can not be found: When I build the project I get the following output
```
[ 91%] Built target Arduino_Drv_HardwareRateDriver
In file included from /home/aggelos/fprime/led-blinker/fprime-arduino/Arduino/Drv/I2cDriver/I2cDriver.cpp:8:
/home/aggelos/fprime/led-blinker/fprime-arduino/Arduino/Drv/I2cDriver/I2cDriver.hpp:11:10: fatal error: Wire.h: No such file or directory
   11 | #include <Wire.h>
      |          ^~~~~~~~
compilation terminated.
gmake[2]: *** [Arduino/Drv/I2cDriver/CMakeFiles/Arduino_Drv_I2cDriver.dir/build.make:93: Arduino/Drv/I2cDriver/CMakeFiles/Arduino_Drv_I2cDriver.dir/I2cDriver.cpp.obj] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from /home/aggelos/fprime/led-blinker/fprime-arduino/Arduino/Drv/SpiDriver/SpiDriverArduino.cpp:8:
/home/aggelos/fprime/led-blinker/fprime-arduino/Arduino/Drv/SpiDriver/SpiDriver.hpp:11:10: fatal error: SPI.h: No such file or directory
   11 | #include <SPI.h>
      |          ^~~~~~~
compilation terminated.
gmake[2]: *** [Arduino/Drv/SpiDriver/CMakeFiles/Arduino_Drv_SpiDriver.dir/build.make:105: Arduino/Drv/SpiDriver/CMakeFiles/Arduino_Drv_SpiDriver.dir/SpiDriverArduino.cpp.obj] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from /home/aggelos/fprime/led-blinker/fprime-arduino/Arduino/Drv/SpiDriver/SpiDriver.cpp:8:
/home/aggelos/fprime/led-blinker/fprime-arduino/Arduino/Drv/SpiDriver/SpiDriver.hpp:11:10: fatal error: SPI.h: No such file or directory
   11 | #include <SPI.h>
      |          ^~~~~~~
compilation terminated.
In file included from /home/aggelos/fprime/led-blinker/fprime-arduino/Arduino/Drv/I2cDriver/I2cDriverArduino.cpp:8:
/home/aggelos/fprime/led-blinker/fprime-arduino/Arduino/Drv/I2cDriver/I2cDriver.hpp:11:10: fatal error: Wire.h: No such file or directory
   11 | #include <Wire.h>
      |          ^~~~~~~~
compilation terminated.
gmake[2]: *** [Arduino/Drv/SpiDriver/CMakeFiles/Arduino_Drv_SpiDriver.dir/build.make:91: Arduino/Drv/SpiDriver/CMakeFiles/Arduino_Drv_SpiDriver.dir/SpiDriver.cpp.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:4922: Arduino/Drv/SpiDriver/CMakeFiles/Arduino_Drv_SpiDriver.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake[2]: *** [Arduino/Drv/I2cDriver/CMakeFiles/Arduino_Drv_I2cDriver.dir/build.make:107: Arduino/Drv/I2cDriver/CMakeFiles/Arduino_Drv_I2cDriver.dir/I2cDriverArduino.cpp.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:4882: Arduino/Drv/I2cDriver/CMakeFiles/Arduino_Drv_I2cDriver.dir/all] Error 2
[ 92%] Built target Svc_ComStub
[ 93%] Built target Svc_Deframer
[ 94%] Built target Svc_Framer
gmake: *** [Makefile:136: all] Error 2
[ERROR] CMake erred with return code 2
```

I've checked many times the repository and my code if it has all the includes and I'm sure I have same makefiles (I cloned the fprime-arduino repo). 

I should also note that I haven't found any related issues, discussions or notes about it in tutorials

aggelosmots, 2024-03-12:
Looks like I needed to modify the cookie cutter line 29 like in fpp-tools cookie cutter `{{"Svc::RateGroupDriver::DividerSet rateGroupDivisors{{{1, 0}, {2, 0}, {4, 0}}};"}}`  then it builds successfully. 

The arduino cookie cutter repo has `Svc::RateGroupDriver::DividerSet rateGroupDivisors = {{{100, 0}, {200, 0}, {1000, 0}}};` and if I replace it with `:` it generates `Svc::RateGroupDriver::DividerSet rateGroupDivisors = ({100: 0}, {200: 0}, {1000: 0});`

ethancheez, 2024-03-13:
Thank you for catching this. Mistake on my part with the cookiecutter and will fix it right away.

This is the correct cookicutter syntax for that line you mentioned
```
{{"Svc::RateGroupDriver::DividerSet rateGroupDivisors{{{100, 0}, {200, 0}, {1000, 0}}};"}}
```

ethancheez, 2024-03-13:
Cookiecutter should be fixed.

I also bumped this repo to fprime v3.4.3, which includes a crucial change to the cookiecutter and `fprime-arduino` to support fprime v3.4.3.

closed 2024-03-13 by aggelosmots