fprime-community/fprime-tutorial-arduino-blinker #13: Failed to open GPIO

Nainikas, 2024-06-25:
The tutorial leads the LED to blink and connect to GPIO 13. After configuration and gds, the GPIO fails to open everytime and displays an error on GPIO 6. The RateGroups also display warnings and the parameter file fails to be read.

[image]

[image]

ethancheez, 2024-06-25:
Could you let me know which step you are in the tutorial? And what kind of hardware are you using?

By quick inspection based on the screenshots you provided, the Arduino deployment cookiecutter failed as I see a `LinuxGpioDriver`, which does not work on baremetal devices. Likewise, the log output shows a client-server connection which was not included in the initial tutorial.

The log output shows a "stack size too small" likely because of the comQueue, the depths are too large which will require significant memory that most baremetal devices do not have.

ethancheez, 2024-11-18:
Closing as this seems error seems to relate to the [`Linux Workshop LedBlinker Tutorial`](https://github.com/fprime-community/fprime-workshop-led-blinker)

closed 2024-11-18 by ethancheez