fprime-community/fprime-tutorial-arduino-blinker #14: Build Fails Without Deployment

LeStarch, 2024-07-15:
Following the steps, but building after "new project" and before "new deployment" causes a crash with "cannot find #include <TimeLib.h>"

ethancheez, 2024-07-15:
Did you run:
```sh
arduino-cli lib install Time
```

LeStarch, 2024-07-15:
I did.  It works with building the deployment. Just not the raw project.

LeStarch, 2024-07-15:
It might be that some of the hooks for libraries etc are done on deployments and not on the full project build.

ethancheez, 2024-07-15:
Right. I see what's going on. Yes you are correct. When running `fprime-util build` when no deployment is present (i.e. if you want to build for the Led component itself), the arduino-cli wrapper doesn't seem to be importing all the arduino libraries correctly which result in dependency errors. 

In the tutorial, I ended up putting the [`Add Arduino Deployment`](https://fprime-community.github.io/fprime-tutorial-arduino-blinker/docs/building-for-arduino.html#add-arduino-deployment) step near the beginning before running `fprime-util build`.

ethancheez, 2024-07-15:
FYI, this is not a TimeLib issue. I remember getting a similar error with the SPI and Wire.h libraries, they are also not linked properly when no component/deployment is present in the project.

LeStarch, 2024-07-15:
Yes, this call comes from the line:  `finalize_arduino_executable("${FPRIME_CURRENT_MODULE}")` in the deployment CMake.  This line fixes library usages...but isn't done at the project level.

For now, I am just tracking the issue.