fprime-community/fprime-tutorial-arduino-blinker #15: fprime-util generate but has a error "Could not run 'arduino-cli' "

henrybb0826, 2024-08-22:
Excuse me, I follow the steps by tutorial to the steps of fprime-util generate after creating a new deployment.
However, I got some error saying "Could not run 'arduino-cli' please install 'arduino-cli' and board  'rp2040:rp2040:rpipicow'."
I am pretty sure that I installed arduino-cli and the core of rp2040 in fprime-venv.
Am I missing some path-adding steps or other important things?
[image]
[image]

ethancheez, 2024-08-22:
From your screenshot, I see this:
```
CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles". CMAKE_MAKE_PROGRAM is not set. You probably need to select a different build tool.

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage

CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
```

Do you have a C/C++ compiler installed on your system? And also Make installed?

For example if you type these:
```
g++

make
```
Are these commands found?

henrybb0826, 2024-08-22:
Yes, they are all installed.
[image]
After searching other issues I found that in [this link](https://github.com/SterlingPeet/fprime-arduino/pull/12#issuecomment-1644581188) has the same situation just like me.
However, I don't understand the meaning of the solution of resetting the persistence layer and populating the Arduino workspace directory with the board packages.

ethancheez, 2024-08-22:
@capsulecorplab What was your solution to this issue? Seems like you had a similar issue last year with your VM workspace setup.

capsulecorplab, 2024-08-23:
> @capsulecorplab What was your solution to this issue? Seems like you had a similar issue last year with your VM workspace setup.

I had to use a specific commit from @StirlingPeet's arduino-cli wrapper to fix it in my virtual environment https://github.com/fprime-community/fprime-workspace-image/blob/main/playbook.yaml#L265. You could try doing the same or just pull and run the [fprime-workspace-image](https://github.com/fprime-community/fprime-workspace-image) since it's already configured to build the led blinker project

SterlingPeet, 2024-08-23:
Hello Folks, progress was made on the arduino-cli-wrapper just days ago, I haven’t had the chance to add some tests and push it.  I’ll push the code that appears to be working and link a commit here in the next day or so.

@henrybb0826 are you able to compile an arduino sketch with the cli, and target the rpipocow board?  This would confirm that it is installed and that the next step is try my code that is not yet available.